adafruit/Adafruit_BME280_Library #6: Missing feature, sleep mode.

liba2k, 2016-03-03:
Hi,
Can you please implement a way to send the BME to sleep mode?
Something like:

void Adafruit_BME280::end() {
  write8(BME280_REGISTER_CONTROL, 0);
}

Thank you,
Itai.

mozzbozz, 2016-08-17:
Related: https://github.com/adafruit/Adafruit_BME280_Library/issues/8

solars, 2017-12-28:
is this still open? I notice that sparkfun already has this for quite some time so I might switch

Elfe, 2019-01-05:
in my code I reset the chip after reading the data to reduce power usage
without reset the esp32 + sensor is at about 350µA with the ESP32 in deep sleep
with reset the esp32 + sensor is at about 12µA

a public reset function would be nice

read stuff...
I2Cbus.beginTransmission(BMX280_ADDRESS);
I2Cbus.write(0xE0);
I2Cbus.write(0xB6);
I2Cbus.endTransmission();
sleep

hoffmannjan, 2019-06-20:
this functionality is fully tested in master now :). thanks guys!

closed 2019-06-20 by hoffmannjan

solars, 2019-06-23:
@hoffmannjan is this documented somewhere?

hoffmannjan, 2019-06-23:
@solars I'll add proper doxygen in this repo next week probably, but for now there is some hints in the code 
https://github.com/adafruit/Adafruit_BME280_Library/pull/13/files#diff-35e01fdd369e009c1e031cd9e83dac0bR169

sergerold, 2021-06-07:
> @solars I'll add proper doxygen in this repo next week probably, but for now there is some hints in the code
> https://github.com/adafruit/Adafruit_BME280_Library/pull/13/files#diff-35e01fdd369e009c1e031cd9e83dac0bR169
> and you can also look at whole comments section at #13

Please could someone confirm which method to use for sleep mode? Should I be setting MODE_FORCED? If so, where?

Thanks