adafruit/Adafruit_BME280_Library #8: Missing feature - Oversampling, Sensor Mode, Standby Time, IIR Filter

camrex, 2016-03-16:
Is it possible to add the ability to adjust the following settings:

- Sensor mode (Sleep, Forced, Normal)
- Standby time (0.5ms - 1000ms)
- Humidity Oversampling (x1 to x16)
- Pressure Oversampling (x1 to x16)
- Temperature Oversampling (x1 to x16)
- IIR Filter (off, 2, 4, 8, 16)

From what I can tell, it is currently set for Normal mode with 0.05ms standby, and x16 oversampling for humidity, pressure, and temperature. This is overkill for use as a weather sensor (which Bosch recommends settings of (Forced, 1 sample, x1 oversampling [hum,pres,temp], IIR filter off)

Thanks.

ladyada, 2016-03-16:
if you submit a pull request, we'll look and merge it!

closed 2016-05-06 by camrex

julien-lebot, 2016-05-08:
I've made a pull-request with the features requested if anyone is interested: #9