adafruit/Adafruit_BME280_Library #15: I2C address for BME280 GYBMEP

techi602, 2016-10-05:
Thanks for awesome library. However I had some hard time to make my sensor working. 
The default I2C address is set to 0x77 - (which is also default for BMP180)
I have noticed there are several variants of the sensor

Mine however is running on 0x76 address (many thanks to I2C scanner)

[image]

Means that this library works fine when changing the default I2C address
I have also found this reference in bosch library https://github.com/BoschSensortec/BMP280_driver/blob/master/bmp280.h
as BMP280_I2C_ADDRESS1

Perhaps I would create PR which allows to set I2C address in constructor ( with commented macro i.e. BME280_ADDRESS_GYBMEP )

not sure how common issue is it but I believe someone else may appreciate it

golvellius, 2016-10-18:
Thanks a lot techi602 ...

It's very usefull to me ...

Deliaz, 2016-11-21:
Thanks, it helped me.

rogerclarkmelbourne, 2016-11-30:
Similar issue applies to the BMP280 lib as well

Bruce17, 2016-12-01:
Thanks! I have the same issue with the same sensor 😉

ladyada, 2016-12-01:
yes, the code supports passing in alternate i2c addresses already :)

https://github.com/adafruit/Adafruit_BME280_Library/blob/master/Adafruit_BME280.h#L133

closed 2016-12-01 by ladyada

Bluebrain2000, 2017-03-30:
Thanks! Helped me too!

DooMMasteR, 2017-04-24:
Bu the way, it is pretty easy to change the address.
Of the 3 solder pads the first 2 are connected, rip the connection with a knife and connect the center one with the last pad instead, this will pull the SDO pin of the BME280 up and change the base address to 0x77 which is convenient if you want to use 2 BME280 on the same bus.

Bender-Ender, 2017-06-01:
>Of the 3 solder pads the first 2 are connected, rip the connection with a knife and connect the center one with the last pad instead

Thanks DooMMasteR. I thought that might be the case but couldn't find anywhere that said that conclusively.

sruthikrahulk, 2018-03-14:
Did i get the lpc1768  code for BMP280

kukyarce, 2018-04-15:
with this  
https://github.com/Seeed-Studio/Grove_BME280
works with no traumatic  board work
https://www.youtube.com/watch?v=PidE-pJiIXY

Povilasgtar, 2018-08-07:
Thank you for a link to Seeed, really useful, just worth mentioning, that I am using a BMP sensor attached to a GY-91 module instead of a BME sensor, but it has the same address as BME (0x76), so all that I had to do is just download and install the BMP280 zip also change the address entry in the library file in the sketch folder and everything works perfectly!

Coconaut11, 2018-11-26:
Man, thank you so fvcking much. Honestly without this thread I was done. Thanks.

JasonG-FR, 2018-12-16:
For anyone having the same issue, here is how I managed to make it work :
I simply replaced `status = bme.begin();` by `status = bme.begin(0x76);` in the bme280test.ino sketch.

877dev, 2019-02-24:
@JasonG-FR thanks for the comment, it really helped me!
Was having trouble getting two sensors to read in i2c, I needed to use:

bool status;

status = bme1.begin(0x77);
status = bme2.begin(0x76);

Cheers :)

ColomeDisco, 2019-07-01:
Hi, Is possible to connect 4 devices at the same i2c bus? 
I want to make a project to diferential values like a pitot Tube
tanks

JasonG-FR, 2019-07-01:
@ColomeDisco yes it's possible to connect more than 4 devices on the i2c bus. 

The issue is to get each one a unique address. If you want to use 4 BME280, the Adafruit ones have 2 jumper solder pads that could be used to change their address, so 4 unique addresses in total.

ColomeDisco, 2019-07-01:
Ok, Thanks for the response, but I not have clear documentation how to do it...
do you have some link or information about it? 
thanks again!!

JasonG-FR, 2019-07-01:
@ColomeDisco Sorry I made a mistake, you can only connect 2 Adafrui module on the i2c bus at the same time : https://forums.adafruit.com/viewtopic.php?f=22&t=128043
If you want to use 4 at the same time, you could use SPI:

> If you want to connect multiple BME280's to one microcontroller, have them share the SDI, SDO and SCK pins. Then assign each one a unique CS pin.

from https://learn.adafruit.com/adafruit-bme280-humidity-barometric-pressure-temperature-sensor-breakout/pinouts

bigjack48, 2020-08-02:
For who has the same issue and search for a quick solution:

simply use `bme280 = adafruit_bme280.Adafruit_BME280_I2C(i2c, address=0x76)`

from https://circuitpython.readthedocs.io/projects/bme280/en/latest/

dan0delellis, 2021-01-24:
I was getting `OSError: [Errno 121] Remote I/O error` and `ValueError: No I2C device at address: 77` after my sensor made contact with condensation in a gravity-assisted experiment failure.  This cleared the errors right up!

joaofgoncalves, 2024-05-05:
Worked for me as well with bme.begin(0x76)