adafruit/Adafruit_BME280_Library #21: Problem with ESP8266

pilnikov, 2017-02-21:
The new version of the library (1.0.5) does not work with my Chinese BME280 on ESP8266. Whereas 1.0.4 works without problems. I change the address 0x76 only.
In indications i see  zeros only.

ladyada, 2017-02-28:
just tried it with an adafruit bme280 and a huzzah esp8266 and it works fine?

closed 2017-02-28 by ladyada

pilnikov, 2017-02-28:
I live in Russia. I tried to order from your branded stuff (adafruit bme280 and a huzzah esp8266) I did not succeed. Therefore, I have only Chinese BME280 and Wemos D1 mini. It's spring here. And both my Chinese BME280 flooded with water. They are "choked". I'm waiting for the new BME280. When I get - I will describe in more detail - for some reason does not work. I tried to use the .h file of the new version and the old .cpp - so works.

pilnikov, 2017-04-03:
Hello. I got new sensors and checked the work of your library. At the first pass of the program the NAN is issued. From the second pass everything is normalized. I solved this problem by adding a little pause 
(delay (100)) to the end of the begin() method.

pilnikov, 2017-04-03:
`    readCoefficients(); // read trimming parameters, see DS 4.2.2

    setSampling(); // use defaults
    
    delay(100); // Added pause

    return true;
}
`

rdts, 2017-12-25:
I encountered the same issue after updating from 1.04 to 1.06. Adding a delay of 100ms after bme.begin() provided a dirty workaround, but wasn't really a clever solution for my code since I use two sensors and switched between them by calling bme.begin(address) periodically. I ended up creating two instances of Adafruit_BME280.

deanm1278, 2017-12-27:
hi @rdts, are you using an ESP8266 as well? If so, what version of the esp8266 core are you using (check the boards manager). Can you post the code you are using?
I just tested the bme280test example on an Adafruit Feather Huzzah ESP8266 with core version 2.3.0 and don't seem to be having any issues.

deanm1278, 2017-12-27:
ok @rdts after looking into this a bit it seems like that delay is necessary for the first sample to be ready. I have added it to the end of Adafruit_BME680::init()

rdts, 2017-12-28:
@deanm1278  Thanks, your commit should fix this. I guess this was only an issue for people who call begin() frequently from inside the loop.