adafruit/Adafruit_BME280_Library #29: ESP32 hangs after loading the lib

micropet, 2017-11-02:
The current version no longer works on an ESP32 board.

The code compiles without errors, the ESP hangs after the lib is loaded.
version = 1.0.5 runs without problems.

AlexSwensen, 2017-11-02:
@micropet posting some example code is helpful.

luckyhacky, 2017-12-19:
I think i is caused either by line 103 or 343 of the source 
https://github.com/adafruit/Adafruit_BME280_Library/blob/master/Adafruit_BME280.cpp#L103
```
        while (isReadingCalibration())
        delay(100);
```

https://github.com/adafruit/Adafruit_BME280_Library/blob/master/Adafruit_BME280.cpp#L343
```
        while (read8(BME280_REGISTER_STATUS) & 0x08)
        delay(1)
```

Exceptions in reads, e.g. if read is not ack (I2C) or 0x00 is returned (SPI), there will be a neverending loop.
Normally it should be limited or a timeout introduced.

Solution could be:
```
int timeout == 42;
while (isReadingCalibration() && timeout > 0)
{
        delay(100);
        timeout --;
}
```

Nevertheless please provide code snippet and check connections.

ladyada, 2019-12-29:
closing due to lack of activity

closed 2019-12-29 by ladyada