adafruit/Adafruit_BME280_Library #31: have 2 BME280 by 0x76 and 0x77 simultaneously side by side on same i2c Bus?

tofrnr, 2018-01-20:
how is it possible to have 2 BME280 by 0x76 and 0x77 simultaneously side by side on same i2c Bus?
For that purpose the ability to create instances by passing the bus address would be helpful and then wiring CS to GND or Vc accordingly:
```
#include <Adafruit_Sensor.h>
#include <Adafruit_BME280.h>

Adafruit_BME280 bme1(0x76);  
Adafruit_BME280 bme2(0x77);  
```

will that be provided?

ladyada, 2018-01-20:
addresses are called in to begin()
https://github.com/adafruit/Adafruit_BME280_Library/blob/master/Adafruit_BME280.cpp#L50

closed 2018-01-20 by ladyada

tofrnr, 2018-01-21:
yes, but as stated: I need 2 adresses, simultaneously, for 2 devices! So how to have 2 arbitrary instances simultaneously?

tofrnr, 2018-01-22:
why do you close this issue? 
it is not solved - how shall I call begin with 2 different device addresses??

andrei-ivanov, 2018-01-22:
Probably because it's not a library issue, but a general coding issue :-)

tofrnr, 2018-01-22:
so what is the solution - I don't see a chance to pass an address to begin() by value

andrei-ivanov, 2018-01-22:
You were given a link to the method that allows to do that.
If you don't know how to invoke it, please try a coding guide.

tofrnr, 2018-01-22:
I don't understand - in the link there is just the lib source code, whtal am I expected to do with that?
I want to use the lib, with 2 different addresses for 2 different instances, so what to do?
If possible, show me, f not, it's a lib issue and the lib has to be improved.

andrei-ivanov, 2018-01-22:
It's possible, just use Google