adafruit/Adafruit_BME280_Library #34: Error running example

MatejKovacic, 2018-02-15:
Hi,

I have Arduino Nano (ATMega 328P) and sensor BME/BMP280.

I try to run example advancedsettings.ino and got this error (I am using Arduino IDE):
```
/home/matej/Arduino/Adafruit_BME280_Library/advancedsettings/advancedsettings.ino:21:29: fatal error: Adafruit_BME280.h: No such file or directory
 #include <Adafruit_BME280.h>
                             ^
compilation terminated.
exit status 1
Error compiling for board Arduino Nano.
```

So I replaced (in file advancedsettings.ino) this line:
`#include <Adafruit_BME280.h>`
With this line:
`#include </home/matej/Arduino/Adafruit_BME280_Library/Adafruit_BME280.h>`

Now I get this error:

```
/home/matej/Arduino/Adafruit_BME280_Library/advancedsettings/advancedsettings.ino: In function 'void setup()':
advancedsettings:40: error: 'Wire1' was not declared in this scope
     if (! bme.begin(&Wire1)) {
                      ^
exit status 1
'Wire1' was not declared in this scope
```

Any idea what could be wrong?

ladyada, 2018-02-15:
its definitely something with your setup
did you purchase the BME sensor from adafruit?

MatejKovacic, 2018-02-16:
No, I have bought a Chinese clone from AliExpress. Device is marked as "BME/BMP280". Anyway, I have found the  problem, my I2C device address is 0x76. Now it is working.

Anyway, thanks for your great tutorials and things you are doing.

closed 2018-02-16 by ladyada