adafruit/Adafruit_BME280_Library #36: Error: 'Wire1' was not declared in this scope in  "advancedsettings" example from the Adafruit BME280 library

GeoffLethbridge, 2018-02-25:
- Arduino board:  UNO

- Arduino IDE version (found in Arduino -> About Arduino menu):  1.8.5

The "advancedsettings" example from the Adafruit BME280 library gives the following error:   
/----------------------------------
exit status 1
'Wire1' was not declared in this scope
/----------------------------------
Removing 'Wire1' ie bme.begin(&Wire1) -> bme.begin(), fixes the error and the example compiles OK.

themanfrommoon, 2018-06-02:
Hi, I have the same Problem?!

nopnop2002, 2019-05-18:
Remove esp32 from board manager
Delete C:\Users\your_user\AppData\Local\Arduino15\packages\esp32
Install esp32 from board manager again.

hoffmannjan, 2019-06-20:
Looks like the solution is well known, closing this.

closed 2019-06-20 by hoffmannjan