adafruit/Adafruit_BME280_Library #37: More helpful docs for esp32 and i2c

xaque208, 2018-03-01:
Hi,  In trying to use this library with a new esp32 dev board, I'm unable to read a connected bme280.  This library says i2c just works, but nothing about which pins to use, and maybe this is my issue.

I'd like to be able to use the example to know if I can read the data from the sensor.

I'm currently using Wire 22 and Wire 21 from this diagram.  Is this correct?  Icant get the code past the wiring check.  Please advise.

https://github.com/espressif/arduino-esp32/blob/master/docs/esp32_pinmap.png

Cheers.

ladyada, 2018-05-22:
this isnt a library issue, its a hardware issue. please contact your ebay seller for support

closed 2018-05-22 by ladyada

xaque208, 2018-05-22:
@ladyada I think I purchased the BME280 from Adafruit.  @yoursunny got theirs from Ebay.  I have issues with reliability when codiing.  Sometimes, the I2C device is not detected, and sometimes it is.  When the code does run, sometimes it just stops at some point and I have to restart the device.

ladyada, 2018-05-22:
xaque - you can post in the adafruit forums for tech support but we know that esp32 i2c has some bugs and is not stable as other chipsets :)

xaque208, 2018-05-22:
Okay cool.  I'll have a look over the forums.  Thanks @ladyada.  There is also https://github.com/espressif/arduino-esp32/issues/741

MartinHannah, 2019-03-22:
> I have Heltec WiFi_Kit_32 and a cheap BME280 from eBay.
> I2C pins are [21,22](https://github.com/espressif/arduino-esp32/blob/dd639c4a029b3b31559bb4243a066e2005bdbc18/variants/heltec_wifi_kit_32/pins_arduino.h#L22-L23). Since ESP32 implements I2C in hardware, there's no need to specify in library.
> [i2c_scanner](https://playground.arduino.cc/Main/I2cScanner) says the I2C address of my sensor is 0x76 instead of 0x77. Successful scanner return also indicates hardware wiring is correct.
> The problem seems to be
> 
> [Adafruit_BME280_Library/Adafruit_BME280.cpp](https://github.com/adafruit/Adafruit_BME280_Library/blob/4a84ebc99e5fbda1050a3c4843ad349ed60d53ca/Adafruit_BME280.cpp#L92-L93)
> 
> Lines 92 to 93 in [4a84ebc](/adafruit/Adafruit_BME280_Library/commit/4a84ebc99e5fbda1050a3c4843ad349ed60d53ca)
> 
>  if (read8(BME280_REGISTER_CHIPID) != 0x60) 
>      return false; 
> that asserts chipID must be 0x60.
> The other guy's library [allows both](https://github.com/sparkfun/SparkFun_BME280_Arduino_Library/blob/fdc9cd3b08d04730e8e62a4784bfe92f9c804922/src/SparkFunBME280.cpp#L116-L118) 0x60 and 0x58, and my sensor is detected correctly.
> (although the readings are all wrong, and I'm still investigating)

This is the missing information I have been searching for thanks.
I have been trying to get ESP32 I2C to work reliably for days, trying different boards, I2C pins the works.
Not when I remove this line in Adafruit_BME280.cpp it all works..
So you need to make sure you have done the following to et I2C working on ESP32 :
a) Use the correct wiring (duh)
b) Use the default I2C pins SDO_pin 22 and SCK_pin 21
c) status = bme.begin(0x76); 
d) Comment out lines in Adafruit_BME280.cpp
    //if (read8(BME280_REGISTER_CHIPID) != 0x60)
    //    return false;
==============================================================
ALSO: Tested using TTGO SX1276 LoRa ESP32 Bluetooth WI-FI Internet
The BME280 is an important 3 in one sensor for Lora use-cases and it needs to use I2C because on TTGO and Heltec boards (I am evaluating) SX1276 and OLED displays use other pins.ESP

yoursunny, 2019-03-22:
I have several authentic Adafruit BME280 boards now. This library works unchanged.

The unit from eBay appears to be BM**P**280 and it never worked. I got a refund.