adafruit/Adafruit_BME280_Library #42: Handling BMP280?

szilvasyz, 2018-09-20:
Dear Friends,
Is there any chance to enhance this library to handle BMP280 sensor (without humidity measuring) too? If you think I can offer my contribution in this enhancement.
Zoltan

ladyada, 2018-09-20:
no, we have a separate library for that :)

closed 2018-09-20 by ladyada

szilvasyz, 2018-09-20:
Ok, I see, if you thought about Adafruit_BMP280 (https://github.com/adafruit/Adafruit_BMP280_Library). But that library is 1.5 years old and not a drop-in replacement of BME280 lib. Difference is not only lack of humidity-related functions, but I could not found no setSampling() nor takeForcedMeasurement() nor anything about mode settings. These would be great in a low-power, very low sampling rate (for example weather monitoring) application and all is already implemented in BME280 library.
As I think, the simplest way to modify BME280 lib to handle BMP280 too would be enabling to open devices with IDs either 0x60 or 0x58, perhaps along with an ID-query function provided to the user. Then the user would have the possibility for proper handling humidity-related data.

ladyada, 2018-09-20:
OK! please add any pull requests to that library - thats a better place for them :)