adafruit/Adafruit_BME280_Library #50: License?

kankaristo, 2019-04-26:
The repository doesn't have a LICENSE or COPYING file, which license is the library distributed under?

If it's up for a vote, I'd suggest MIT or some other permissive license.

kankaristo, 2019-04-26:
Just noticed that the source files mention the BSD license.

I'll leave this issue open for now, since most people probably look for the license in either a file called LICENSE or COPYING.

siddacious, 2019-04-27:
@kankaristo thanks for the feedback; If you wanted to submit a PR I'd be happy to review it

kankaristo, 2019-04-27:
I'd be happy to make a PR, but there are actually a few different versions of the BSD license, and the license mention in the source files don't specify which one:
- 2 clause BSD
- 3 clause BSD ("New BSD")
- 4 clause BSD ("Original BSD")

The 4 clause BSD license is considered "problematic", because it includes a clause that all marketing material must include this sentence verbatim:
> This product includes software developed by the University of California, Berkeley and its contributors.

So, that leaves the 2 clause and the 3 clause licenses. The only difference between those licenses is this clause:
> Neither the name of nor the names of its contributors may be used to endorse or promote products derived from this software without specific prior written permission.

The 3 clause license is probably more common, and if you ask for permission to "promote" the fact that your software uses Adafruit libraries, the 2 and 3 clause licenses are identical. And the line between "promotion/endorsement" and simple "acknowledgement" is up for debate.

I'll make a PR for the 3 clause BSD license + mention of the LICENSE file in the source files.

kankaristo, 2019-04-27:
PR created:
https://github.com/adafruit/Adafruit_BME280_Library/pull/51

siddacious, 2019-04-28:
closed via PR #51

closed 2019-04-28 by siddacious