adafruit/Adafruit_BME280_Library #52: Compile error in example bme280test

nim, 2019-05-21:
I have compile error, than I try to compile BME280 example in this repo.

[image]
[image]

I have instaled library:
[image]

I also tried manual include:
[image]

caternuson, 2019-05-21:
It looks like you are using an updated and unreleased example that will not work with the current release version of the library.

How did you get to that version of the example? Via the Arduino IDE? From copy/pasting from github? Or something else?

nim, 2019-05-23:
Yes, I used the current example from Github. You need to make examples so that they contain an explicit indication of the head files. At first I tried to run without libraries, and there was another problem. For beginners, it is very important that the example works without additional effort. Following this rule, your library will have many more users.

caternuson, 2019-05-23:
Use the Arduino Library Manager to install the latest release version, including the examples:
**Sketch -> Include Library -> Manage Libraries...**
And then get at the examples via the Arduino IDE also:
**File -> Examples ->**

The release versions of everything should work. By grabbing the current version directly from this repo, you are using updates that haven't been officially released yet.

ladyada, 2019-05-23:
hiya if you follow
https://learn.adafruit.com/adafruit-bme280-humidity-barometric-pressure-temperature-sensor-breakout/arduino-test#install-adafruit-bme280-library-4-6
you should be good to go! current version passes travis, so closing.

closed 2019-05-23 by ladyada