adafruit/Adafruit_BME280_Library #58: BME280.begin() does not support specifying custom I2C pins

lnlp, 2019-08-18:
Adafruit BME280.begin() does not support specifying custom pins for I2C. 
(While for SPI specifying custom pins is supported.)

ESP32 supports 2 hardware I2C interfaces but BME280 library can only be used for the default I2C pins because custom I2C pins cannot be specified.

Please add an option for specifying custom I2C pins when calling BME280.begin().

lnlp, 2019-08-18:
After further investigation this can be solved without a change to BME280.begin().

closed 2019-08-18 by lnlp

stodge, 2022-04-15:
How did you solve this?

caternuson, 2022-04-15:
Check here:
https://github.com/espressif/arduino-esp32/issues/3779

Cristianistrate, 2023-12-23:
> Check here: [espressif/arduino-esp32#3779](https://github.com/espressif/arduino-esp32/issues/3779)

Can you specify exactly how  ?
SetPins is working ?
Thank you.