adafruit/Adafruit_BME280_Library #71: Isn't while(1); dangerous ?

rhuss, 2019-12-14:
Sorry if this is a noob question, just started with tinkering, but I think in the example the `while(1);` loop is quite problematic: https://github.com/adafruit/Adafruit_BME280_Library/blob/616c7290e4a1bf863021506fe8367e924881cc80/examples/bme280test/bme280test.ino#L54

It happened to me that when I tried it with a Wemos LOLIN32 Pro board (an old one, from two years ago), that the board started smoking after some time got very hot when running this example and it didn't detect an I2C device. First I thought that this is common idiom in ESP32 or Arduino coding so that it indeed is translated in a kind of sleep. But now it looks for me that it really just burns CPU cycle, like in any other programming language.

I killed 3 boards that way until I replaced that line with a `delay(verylongtimeout)`.
What's your take on that `while(1);` idiom, which, btw, I have seen several times in various examples (so I'm really curious what's the reasoning behind this loop).

Sorry again if this is a noob question, feel free to close the issue any time. 

- Arduino board:  Wemo Lolin32 Pro (ESP32)

- Arduino IDE version (found in Arduino -> About Arduino menu):  Using plattform.io

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): Upload BME280 example with the sensor being not detected.

caternuson, 2019-12-14:
That is just a simple infinite loop. You should only get to that if the BME280 is not initialized properly for some reason. The idea is to give you an error message and then stop. It should not cause any issues with hardware. The smoking and getting hot must be something else.

closed 2019-12-14 by caternuson

rhuss, 2019-12-15:
Thanks, but that does not explain why a `while(1);` should be different on an ESP32 than on any other system I know which will burn the CPU with 100%.

Try `perl -e 'while(1){}'` on mac or linux machine (but you probably know that ....

ladyada, 2019-12-15:
microcontrollers don't work like that, they are always at 100%