adafruit/Adafruit_BME280_Library #73: Version 2.0.1: Examples do not work any more (i2c)

moeskerv, 2020-01-10:
Recent changes in the initialization of the sensor object results in non working examples:

If I get it right, the latest changed in the begin() method make the I2C addr and the wire object mandatory. As the examples do not provide them the connection to the BME280 will not get set up.

bool Adafruit_BME280::begin(uint8_t addr, TwoWire *theWire)

BR,
Volker

moeskerv, 2020-01-10:
It was caused by this commit:
https://github.com/adafruit/Adafruit_BME280_Library/commit/2a23df09f35986d1fb06a40f107051907ee53fcc

caternuson, 2020-01-10:
The parameters are not required. Defaults are provided in the header file:
https://github.com/adafruit/Adafruit_BME280_Library/blob/master/Adafruit_BME280.h#L218

I just ran the [bme280test](https://github.com/adafruit/Adafruit_BME280_Library/tree/master/examples/bme280test) example and it ran fine. What sketch are you running and how is the board wired?

moeskerv, 2020-01-10:
Oh, yes. I have my BME280 wired to 0x76, the default is 0x77 so that is the reason. 
Thanks a lot for the feedback and sorry for the confusion!

So, all is fine! Have a good weekend.

closed 2020-01-10 by moeskerv