adafruit/Adafruit_BME280_Library #75: Possible hang when using Adafruit_BME280::takeForcedMeasurement()

netmaniac, 2020-02-22:
When sensor BME280 gets disconnected or will is damaged calling Adafruit_BME280::takeForcedMeasurement() can results in hanging. Reason is in loop https://github.com/adafruit/Adafruit_BME280_Library/blob/0213f84f9fe1344b99042f3a8051f489ec1ddc5a/Adafruit_BME280.cpp#L357

On ESP8266 this loop won't trigger WDT, since it is using delay.

To reproduce - connect BME280 sensor to ESP, initialize it in foreced mode and kead in some intervals (10s). Then disconnect sensor from SPI bus, or power it off. ESP will hang.

kimi1991, 2020-09-24:
I'll work on this.

siddacious, 2020-10-31:
fixed! thanks @kimi1991 

@netmaniac let us know if this isn't able to fix your issue

closed 2020-10-31 by siddacious