adafruit/Adafruit_BME280_Library #84: Custom Temperature Adjustments

AgentZombie, 2020-08-01:
Preemptive grain of salt: I'm pretty new to electronics so if anything I say here doesn't quite make sense, that's why.

I have a handful of Chinese white-label BME280 boards all of which report a temperature that's slightly off, one as much as +3C. I understand that the temperature reading is intended as an estimate and not a true value.

In my own code I can adjust the temperature I report by the required amount. However, the incorrect temperature is being used in pressure and humidity calculations, making them less accurate.

Would it be reasonable to specify a temperature adjustment for the sensor object or to be supplied to the humidity and pressure calculation functions to account for the discrepancy? Should I just commit the grossly incorrect sensors to the e-waste bin?

AgentZombie, 2020-08-01:
[image]

I have a sensor outside and one in my living room. The attached graph shows the difference in pressure which seems to indicate I keep a negative-pressure house :laughing:.

There's are two regions with no samples where I moved the living room sensor outside for comparison and to capture the difference in temperature error between a cool house and hot outdoors. The graph shows that the difference in pressure continues even though while outside the two sensors were within a meter of each other.

I'm speculating that the error in temperature reading is being propagated to the calculated pressure. I suspect that supplying a correction for the inaccurate temperature would also make the pressure and humidity more accurate.

ladyada, 2020-08-02:
hi feel free to submit a PR, we have no plans to add it - the affect may be negligable

closed 2020-08-02 by ladyada