adafruit/Adafruit_BME280_Library #86: Inaccessible class member functions

tho1010, 2020-08-14:
- Arduino board:  board independent software bug
- Arduino IDE version (found in Arduino -> About Arduino menu):  Windows 1.8.13

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): 
the two class member functions
- float Adafruit_BME280::getTemperatureCompensation(vid){}
- void Adafruit_BME280::setTemperatureCompensation(float adjustment){}

are not accessable from a Sketch and generate a compiler error:
'class Adafruit BME280' has no member named 'getTemperatureCompensation'
Despite the fact, that this member is explicitely specified in Adafruit_BME280.cpp and referenced in the public section of Adafruit_BME280.h

Maybe the compiled library code is not identical to the documented code.

ladyada, 2020-08-14:
@AgentZombie can help you with that

AgentZombie, 2020-08-14:
When I submitted the PR it didn't occur to me that the version might need to be bumped in `library.properties`. Before and after my changes are both `2.0.2`. Testing it on a fresh project in platformio shows the issue described unless I specify the dependency as the repo instead of the name.

I'm not C++ savvy and I'm new with MCU dev. My best guess is that `library.properties` needs the version bumped to `2.0.3`. Can someone more experienced weigh in?

caternuson, 2020-08-14:
Yah, looks like it just needs a version bump and a release.

ladyada, 2020-08-14:
we do a sweep once a week and bump old libraries - @siddacious will get to it shortly!

siddacious, 2020-11-23:
released!

closed 2020-11-23 by siddacious