adafruit/Adafruit_BME280_Library #106: Adafruit BME280 library does not work on Raspberry Pi Pico - runtime crashes

sallevan, 2022-12-23:
- Arduino board:  **Raspberry Pi Pico**

- Arduino IDE version (found in Arduino -> About Arduino menu):  **2.0.3**

- Steps to reproduce:

1. make some kind of meteosensor board attaching BME280 via I2C level shifter to Raspberry Pi Pico pins 6 and 7;
2. install Arduino IDE from official site;
3. install Raspberry Pi Pico runtime environment from boards manager;
4. install Adafruit BME280 library with its dependencies via library manager;
5. install DFRobot BME280 library with its dependencies via library manager;
6. compile and upload official demo sketch from Adafruit (f.e, bme280_unified);
7. compile and upload official demo sketch from DFRobot (f.e, raedDataI2C).

If you make board correctly (I did :), both sketches, in theory, should work. Really, **Adafruit's just crashes immediately signalling SOS via built-in LED while DFRobot's work like a charm.**

sallevan, 2022-12-23:
Just to test I've changed Core to [alternate one](https://github.com/earlephilhower/arduino-pico) - LED does not show anything, but serial output does not show anything too. So, I consider that sketch hang.

caternuson, 2022-12-23:
Please post in the forums with photos of your setup showing how everything is connected:
https://forums.adafruit.com/

Could not recreate. Running the `bme280test` example from library on a Pi Pico gives expected output.

[image]

[image]

sallevan, 2022-12-24:
It was my error, I have put sensor address into constructor arguments instead of begin() method.
It might be questionable why library does not check SPI pin given in constructor for validity (I know no microcontroller with pin number 0x76, it must have about 140 pins for it - power, ground, etc...).

closed 2022-12-24 by sallevan