adafruit/Adafruit_BME280_Library #112: Sensor sanity check

baranyaib90, 2024-09-25:
Arduino board:  Arduino Nano v3
Arduino IDE version (found in Arduino -> About Arduino menu):  2.3.2

Dear Maintainers!
I would like to ask for some improvement of this library to make it more robust in case of wireing issues.
Because of thermal expansion sometimes my wires get loose and I have contact issues. I know, this is my problem, I should have solder it, and so on.
But I would expect to have some sanity check functionality from the library side.

Given the following test scenario:

1. Running the code of: [bme280test.ino](https://github.com/adafruit/Adafruit_BME280_Library/blob/master/examples/bme280test/bme280test.ino)
```
BME280 test
-- Default Test --

Temperature = 28.53 °C
Pressure = 1003.00 hPa
Approx. Altitude = 85.73 m
Humidity = 50.89 %

Temperature = 28.74 °C
Pressure = 1003.03 hPa
Approx. Altitude = 85.46 m
Humidity = 50.77 %
```

2. When I disconnect the sensor, I do not get NAN values (except Approx. Altitude), instead invalid values:
```
Temperature = 179.55 °C
Pressure = -160.23 hPa
Approx. Altitude = nan m
Humidity = 100.00 %

Temperature = 179.55 °C
Pressure = -160.23 hPa
Approx. Altitude = nan m
Humidity = 100.00 %
```

3. After reconnecting the sensor, finally there are NAN values:
```
Temperature = nan °C
Pressure = nan hPa
Approx. Altitude = nan m
Humidity = nan %
```

Thanks to NAN values, I can call the begin() function again and I can make the sensor work properly.
But I have no workaround for now to avoid the invalid values.
If there would be a possibility to check if the sensor is really connected and responding properly, that would be great!
For example:

1. sensorID() function could actually querry the sensor ID and not returning a cached value. I don't want to call begin() all the time before measurement.
2. There could be a isConnected() or similar function which checks if the sensor responds to a command with valid data.

Thank you in advance!

LeunyEumel, 2024-09-30:
I am facing the same Issue - or have the same feature request

trying to run two bme280 with I2C over long cables frequently the sensor readings got stuck on:
Humidity:           100%
Temperature:    180.59°C
Pressure:          -157.83
I assume this is due to the I2C erroring

I would prefer a cleaner way than to check my readings against above values to decide when to reset the sensors/the I2C

LeunyEumel, 2024-10-07:
Update: the pullups only partially solved the problem.
Every 12 to 24h approximately the sensor still returns invalid readings.

Seems the Issue might be related to the Sensor itself:
https://community.particle.io/t/bme280-sensor-problem/49627/32

The proposed solution of providing 3v3 via a GPIO and power cycling whenever invalid readings has improved my situation.
I will report back after letting it run for some days

runningWithPyro, 2025-04-03:
Any further information on the root cause of this issue?  

Thanks!

baranyaib90, 2025-04-03:
I did not had the time to debug further.
It works with my workaround and I'm fine with that

LeunyEumel, 2025-04-03:
I suspect the rootcasue beeing noise or Errors on the I2C lines - It only seemed to appear for me once I moved from breadboard to my final wiring >50cm of (Shielded CAT6 Ethernet) cables

I still have the Sensor powered by a GPIO allowing me to powercycle it whenever readings become unplausible.
Has been running fine like that now for 6months - i read the sensor every second but only report the average back to Homeassistant every couple minutes - like this I am not even noticing if the sensor is being reset or not
For me that was the only way to have the Sensor working for long periods of time with a cable length of more than a couple centimeters

runningWithPyro, 2025-04-15:
thank you for the reply, and for the power-via-gpio suggestion.

tyeth, 2025-04-30:
Might be worth trying this [https://www.adafruit.com/product/4756](https://www.adafruit.com/product/4756) 

[image]

Adafruit LTC4311 I2C Extender / Active Terminator, designed to improve the signal quality for longer i2c lines.

> "  When the chip is powered and enabled, it will watch the SCL and SDA lines. When it sees them being pulled up through the I2C resistors, it will activate and dump in some current to give it a boost thru to the top power rail. "

 The product page shows nice graphs of the signals (images: fixed then original signal), and mentions running a BME680 on 100ft of ethernet cable with this.

[image]